luxonis/depthai-core #1233: [BUG] {Title of the bug}

Carter12s, 2025-02-04:
**Check if issue already exists**

- Google it (e.g. *error xy* github luxonis depthai)
- Check [troubleshooting](https://docs.luxonis.com/en/latest/pages/troubleshooting/) in documentation.

**Describe the bug**
Sorry if I'm testing stuff before it is ready...

ColorCamera node has internal fault wtih Rvc4 camera on `v3.0.0-alpha.11`

**Minimal Reproducible Example**
Append the MRE to the bug report, [instructions here](https://docs.luxonis.com/en/latest/pages/support/#creating-minimal-reproducible-example)

```python
#!/usr/bin/env python3

import depthai as dai
import time

pipeline = dai.Pipeline()
pipeline.getDefaultDevice().setLogLevel(dai.LogLevel.DEBUG)
cam = pipeline.create(dai.node.ColorCamera)

pipeline.start()

time.sleep(1)
```

**Expected behavior**
ColorCamera node works with rcv4 cameras as it provides functionality still not possible with Camera node (like getting raw full resolution frames).

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Pipeline Graph**

Please also provide a screenshot of your pipeline using the [DepthAI Pipeline Graph](https://github.com/geaxgx/depthai_pipeline_graph).

**Attach system log**
 
```
$ python3 ./log_system_information.py 
{
    "architecture": "64bit ELF",
    "machine": "x86_64",
    "platform": "Linux-5.15.0-131-generic-x86_64-with-glibc2.29",
    "processor": "x86_64",
    "python_build": "default Dec 13 2020 11:40:11",
    "python_compiler": "GCC 9.3.0",
    "python_implementation": "CPython",
    "python_version": "3.8.6",
    "release": "5.15.0-131-generic",
    "system": "Linux",
    "version": "#141~20.04.1-Ubuntu SMP Thu Jan 16 18:38:51 UTC 2025",
    "win32_ver": "",
    "packages": [
        "attrs==21.2.0",
        "autopep8==1.6.0",
        "bcrypt==3.2.0",
        "certifi==2021.5.30",
        "cffi==1.14.5",
        "chardet==4.0.0",
        "charset-normalizer==3.4.1",
        "cryptography==3.4.7",
        "depthai==3.0.0a12.dev0+19be0f8b89f39bef16fd81d47124c4ac15aad3d6",
        "distro==1.5.0",
        "docker==5.0.0",
        "docker-compose==1.29.2",
        "dockerpty==0.4.1",
        "docopt==0.6.2",
        "flake8==3.8.4",
        "idna==2.10",
        "Jinja2==3.1.5",
        "jsonschema==3.2.0",
        "lxml==4.6.3",
        "MarkupSafe==2.1.5",
        "mccabe==0.6.1",
        "numpy==1.24.4",
        "opcua==0.98.13",
        "opcua-client==0.8.0",
        "opcua-widgets==0.5.10",
        "opencv-python==4.11.0.86",
        "paramiko==2.7.2",
        "pip==25.0",
        "pycodestyle==2.6.0",
        "pycparser==2.20",
        "pyflakes==2.2.0",
        "PyNaCl==1.4.0",
        "PyQt5==5.15.4",
        "PyQt5-Qt5==5.15.2",
        "PyQt5-sip==12.9.0",
        "PyQt6==6.7.1",
        "PyQt6-Qt6==6.7.3",
        "PyQt6_sip==13.8.0",
        "pyrsistent==0.17.3",
        "python-dateutil==2.8.2",
        "python-dotenv==0.17.1",
        "pytz==2021.1",
        "PyYAML==6.0.2",
        "requests==2.32.3",
        "setuptools==49.2.1",
        "six==1.16.0",
        "texttable==1.6.3",
        "toml==0.10.2",
        "urllib3==1.26.5",
        "websocket-client==0.59.0"
    ],
    "usb": [
        "NoLib"
    ],
    "uname": [
        "Linux carter-calculon-20 5.15.0-131-generic #141~20.04.1-Ubuntu SMP Thu Jan 16 18:38:51 UTC 2025 x86_64 x86_64"
    ]
}
System info gathered successfully - saved as "log_system_information.json"
```

**Additional context**

Debug logs:
```
$ DEPTHAI_LEVEL=debug python3 ./carter_test.py 
[2025-02-04 09:37:40.521] [depthai] [debug] Python bindings - version: 3.0.0-alpha.12.dev0+19be0f8b89f39bef16fd81d47124c4ac15aad3d6 from  build: 2025-01-26 14:01:55 +0000
[2025-02-04 09:37:40.521] [depthai] [debug] Library information - version: 3.0.0-alpha.12, commit:  from , build: 2025-01-26 14:01:55 +0000, libusb enabled: true
[2025-02-04 09:37:40.529] [depthai] [debug] Initialize - finished
[2025-02-04 09:37:40.573] [depthai] [debug] Resources - Archive 'depthai-bootloader-fwp-0.0.26.tar.xz' open: 1ms, archive read: 51ms
[2025-02-04 09:37:40.803] [depthai] [debug] Resources - Archive 'depthai-device-fwp-976351fda3b70483e37882f989686bf0aa8c604d.tar.xz' open: 1ms, archive read: 281ms
[2025-02-04 09:37:41.034] [depthai] [info] Adding device to the filtered list: DeviceInfo(name=192.168.15.2, mxid=2821073132, X_LINK_GATE, X_LINK_TCP_IP, X_LINK_RVC4, X_LINK_SUCCESS)
[2025-02-04 09:37:41.034] [host] [debug] Device - OpenVINO version: universal
[2821073132] [192.168.15.2] [1738687061.034] [host] [debug] Device - BoardConfig: {"camera":[],"defaultImuExtr":{},"emmc":null,"gpio":[],"imu":null,"logDevicePrints":null,"logPath":null,"logSizeMax":null,"logVerbosity":null,"network":{"mtu":0,"xlinkTcpNoDelay":true},"nonExclusiveMode":false,"pcieInternalClock":null,"sysctl":[],"uart":[],"usb":{"flashBootedPid":63037,"flashBootedVid":999,"manufacturer":"","maxSpeed":4,"pid":63035,"productName":"","vid":999},"usb3PhyInternalClock":null,"uvc":null,"watchdogInitialDelayMs":null,"watchdogTimeoutMs":null} 
libnop:
0000: b9 13 b9 07 81 e7 03 81 3b f6 81 e7 03 81 3d f6 04 bd 00 bd 00 b9 02 00 01 ba 00 be be bb 00 bb
0020: 00 be be be be be be be 00 bb 00 be be bb 00
[2821073132] [192.168.15.2] [1738687061.360] [host] [debug] Schema dump: {"connections":[],"globalProperties":{"calibData":null,"cameraTuningBlobSize":null,"cameraTuningBlobUri":"","leonCssFrequencyHz":700000000.0,"leonMssFrequencyHz":700000000.0,"pipelineName":null,"pipelineVersion":null,"sippBufferSize":18432,"sippDmaBufferSize":16384,"xlinkChunkSize":-1},"nodes":[[0,{"alias":"","id":0,"ioInfo":[[["","frameEvent"],{"blocking":false,"group":"","id":8,"name":"frameEvent","queueSize":8,"type":0,"waitForMessage":false}],[["","isp"],{"blocking":false,"group":"","id":6,"name":"isp","queueSize":8,"type":0,"waitForMessage":false}],[["","still"],{"blocking":false,"group":"","id":5,"name":"still","queueSize":8,"type":0,"waitForMessage":false}],[["","raw"],{"blocking":false,"group":"","id":7,"name":"raw","queueSize":8,"type":0,"waitForMessage":false}],[["","preview"],{"blocking":false,"group":"","id":4,"name":"preview","queueSize":8,"type":0,"waitForMessage":false}],[["","video"],{"blocking":false,"group":"","id":3,"name":"video","queueSize":8,"type":0,"waitForMessage":false}],[["","mockIsp"],{"blocking":true,"group":"","id":2,"name":"mockIsp","queueSize":8,"type":3,"waitForMessage":false}],[["","inputControl"],{"blocking":true,"group":"","id":1,"name":"inputControl","queueSize":3,"type":3,"waitForMessage":false}],[["","inputConfig"],{"blocking":true,"group":"","id":0,"name":"inputConfig","queueSize":3,"type":3,"waitForMessage":false}]],"name":"ColorCamera","parentId":-1,"properties":[185,34,185,32,0,3,0,136,0,0,0,0,0,0,185,3,0,0,0,185,5,0,0,0,0,0,185,5,0,0,0,0,0,0,0,0,0,0,0,0,0,0,185,3,0,0,0,185,3,134,12,0,43,0,134,5,0,2,0,0,130,172,1,18,0,1,0,98,114,105,100,103,0,128,156,1,186,0,255,189,0,255,10,129,44,1,129,44,1,255,255,255,255,255,255,0,136,0,0,240,65,0,136,0,0,128,191,136,0,0,128,191,1,185,4,0,0,0,0,3,3,4,4,4,255,189,0,0,0,136,0,0,128,63,32,32,186,1,1,190]}]]}
[2821073132] [192.168.15.2] [1738687061.360] [host] [debug] Asset map dump: {"map":{}}
[2025-02-04 09:37:41.361] [depthai] [info] Pipeline not logged yet, logging...
[2025-02-04 09:37:43.622] [depthai] [info] Logs sent successfully
[2025-02-04 09:37:43.624] [depthai] [info] Pipeline logs sent to server
[2821073132] [192.168.15.2] [1738687054.558] [ColorCamera(0)] [error] Node threw exception, stopping the node. Exception message: Internal error occured. Please report. commit: 5d1371967432dd4cf1742b56afbf4c669417bcfd | version: 0.0.1 | file: /work/src/gst/rvc4/GstCamRvc4.cpp:959
[2821073132] [192.168.15.2] [1738687064.703] [host] [debug] Will wait for gate: false
[2025-02-04 09:37:44.703] [depthai] [debug] XLinkResetRemote of linkId: (16)
[2821073132] [192.168.15.2] [1738687064.704] [host] [debug] Timesync thread exception caught: Couldn't read data from stream: '__timesync' (X_LINK_ERROR)
[2821073132] [192.168.15.2] [1738687064.704] [host] [debug] Log thread exception caught: Couldn't read data from stream: '__log' (X_LINK_ERROR)
[2821073132] [192.168.15.2] [1738687065.350] [host] [debug] Device closed, 648
```

moratom, 2025-02-04:
Hello @Carter12s !

Thanks for the report, we'll improve the logs here, but RVC4 only supports the new `Camera` node.

See the examples here https://github.com/luxonis/depthai-core/tree/v3_develop/examples/python/Camera.

closed 2025-02-04 by moratom

Carter12s, 2025-02-04:
Thank you for quick response, makes sense and I was afraid that might be the case.

I apologize for asking another question, but the documentation seems to conflict with reality:

Here indicates that "raw" should exist as an output on the camera node:
https://docs.luxonis.com/software/depthai-components/nodes/camera/

[image]

However the actual code in the `v3.0.0-alpha.11` no longer has `raw` as a pre-existing output. Instead it appears I need to use the "requestOutput()" or "requestFullResolutionOutput()". However, both of these appear to not actually give me access to a raw uncompressed frame?

Is there an intended way to get access to a raw uncompressed frame in the new API?

moratom, 2025-02-04:
We don't yet support accessing the raw frames (I assume you mean frames before debayering here) but the feature is in our internal task list.

I'll ask the team internally about what would be a realistic timeline here and get back to you.

Carter12s, 2025-02-04:
That is perfect, and I don't actually need "raw before debayering".

I may just be an idiot and am missing something in the API, but every frame I get appears to have compression artifacts?

moratom, 2025-02-04:
>That is perfect, and I don't actually need "raw before debayering".

In that case `requestFullResolutionOutput()` should give you uncompressed images in full resolution.
Would you be willing to share the images you get out? There is no compression going on, so artifacts are not expected.

Carter12s, 2025-02-04:
Sorry I think I figured out my issue actually (thank you for the help).

I was testing the image sensors performance in a challenging low-light application and did not realize that Sharpness, ChromaDenoise, and Luma Denoise were enabled by default.

The combination of these effects looked very similar to compression artifiacts to me. Once I disabled these the images look as expected and I'm seeing the actual sensor noise.

Carter12s, 2025-02-04:
Example image of what I was trying to get:

[image]

What I was getting before disabling those features:

[image]