luxonis/depthai-core #1234: [BUG] Higher than Expected Latency on RGB feed for Oak 4 D Pro

Carter12s, 2025-02-05:
**Describe the bug**
Higher than expected latency for RGB feed on `Oak 4 D Pro`. Fully acknowledge I'm testing some pre-release stuff, just filing in case this is helpful.

For a 4000x3000 RGB feed RGB888i I'm measuring 700 - 730ms of latency over a 2.5 Gbps direct connection.
For a 4000x3000 RGB Feed of NV12 I'm measuring 380 - 400ms of latency over a 2.5 Gbps direct connection.

The performance is roughly identical between a Python and C++ client.

For reference on a Basler camera with identical resolution I get latency in the 50ms range which is what I would have expected

**Minimal Reproducible Example**
I'm running what I think is a very simple MRE:

``` Python
#!/usr/bin/env python3

import depthai as dai

pipeline = dai.Pipeline()

cam = pipeline.create(dai.node.Camera).build()
outputQueue = cam.requestOutput((4000, 3000), dai.ImgFrame.Type.RGB888i, fps=20, enableUndistortion=False).createOutputQueue()
outputQueue.setBlocking(False)
outputQueue.setMaxSize(1)
configQueue = cam.inputControl.createInputQueue()

pipeline.start()

while pipeline.isRunning():
    frames = outputQueue.tryGetAll()
    for frame in frames:
        t = frame.getTimestamp()
        dt = dai.Clock.now() - t
        print(f"latency= {dt}")
pipeline.stop()
pipeline.wait()
```
Produces:
```
python3 ./carter_test.py 
latency= 0:00:00.236971
latency= 0:00:00.269722
latency= 0:00:00.338914
latency= 0:00:00.412707
latency= 0:00:00.482376
latency= 0:00:00.562229
latency= 0:00:00.631169
latency= 0:00:00.710560
latency= 0:00:00.734799
latency= 0:00:00.708165
latency= 0:00:00.733494
latency= 0:00:00.707215
latency= 0:00:00.732002
```
C++:
```cpp
#include "depthai/depthai.hpp"
#include "depthai/utility/Clock.hpp"

#include <chrono>

int main(int argc, char** argv) {
    dai::Pipeline pipeline;
    auto camRgb = pipeline.create<dai::node::Camera>()->build();
    auto* output = camRgb->requestOutput(std::make_pair(4000, 3000), dai::ImgFrame::Type::RGB888i, dai::ImgResizeMode::CROP, 20, false);
    if(output == nullptr) {
        std::cout << "Error creating output, exiting\n";
        return 1;
    }
    auto outputQueue = output->createOutputQueue();
    outputQueue->setBlocking(false);
    outputQueue->setMaxSize(1);
    pipeline.start();

    while(pipeline.isRunning()) {
        auto imgFrame = outputQueue->get<dai::ImgFrame>();
        auto timestamp = imgFrame->getTimestamp();
        auto now = dai::Clock::now();
        std::cout << "Latency: " << std::chrono::duration_cast<std::chrono::milliseconds    >(now - timestamp).count() << "ms\n";
        cv::imshow("rgb", imgFrame->getCvFrame());
        int key = cv::waitKey(1);
        if(key == 'q' || key == 'Q') {
            pipeline.stop();
            return 0;
        }
    }
    return 0;
}
```

**Expected behavior**
For simply getting a raw, undistorted RGB image across the network I would have expected latency less than 100ms. It is very unclear to me where the delay is coming from.

If I enable logging I can see:
```
[2821073132] [192.168.15.2] [1738712581.347] [Camera(0)] [debug] Camera timings: color_convert: 29ms
[2821073132] [192.168.15.2] [1738712581.440] [XLinkOut(4)] [debug] XLinkOut total took 123ms, processing 123ms, getting_frames 0ms, sending_frames 0ms
```
Which to me indicates that ~150ms of latency could be ocurring, which wouldn't be crazy, but actual latency is ~5x that?

**Attach system log**
```
$ python3 ./log_system_information.py 
{
    "architecture": "64bit ELF",
    "machine": "x86_64",
    "platform": "Linux-5.15.0-131-generic-x86_64-with-glibc2.29",
    "processor": "x86_64",
    "python_build": "default Dec 13 2020 11:40:11",
    "python_compiler": "GCC 9.3.0",
    "python_implementation": "CPython",
    "python_version": "3.8.6",
    "release": "5.15.0-131-generic",
    "system": "Linux",
    "version": "#141~20.04.1-Ubuntu SMP Thu Jan 16 18:38:51 UTC 2025",
    "win32_ver": "",
    "packages": [
        "attrs==21.2.0",
        "autopep8==1.6.0",
        "bcrypt==3.2.0",
        "certifi==2021.5.30",
        "cffi==1.14.5",
        "chardet==4.0.0",
        "charset-normalizer==3.4.1",
        "cryptography==3.4.7",
        "depthai==3.0.0a12.dev0+19be0f8b89f39bef16fd81d47124c4ac15aad3d6",
        "distro==1.5.0",
        "docker==5.0.0",
        "docker-compose==1.29.2",
        "dockerpty==0.4.1",
        "docopt==0.6.2",
        "flake8==3.8.4",
        "idna==2.10",
        "Jinja2==3.1.5",
        "jsonschema==3.2.0",
        "lxml==4.6.3",
        "MarkupSafe==2.1.5",
        "mccabe==0.6.1",
        "numpy==1.24.4",
        "opcua==0.98.13",
        "opcua-client==0.8.0",
        "opcua-widgets==0.5.10",
        "opencv-python==4.11.0.86",
        "paramiko==2.7.2",
        "pip==25.0",
        "pycodestyle==2.6.0",
        "pycparser==2.20",
        "pyflakes==2.2.0",
        "PyNaCl==1.4.0",
        "PyQt5==5.15.4",
        "PyQt5-Qt5==5.15.2",
        "PyQt5-sip==12.9.0",
        "PyQt6==6.7.1",
        "PyQt6-Qt6==6.7.3",
        "PyQt6_sip==13.8.0",
        "pyrsistent==0.17.3",
        "python-dateutil==2.8.2",
        "python-dotenv==0.17.1",
        "pytz==2021.1",
        "PyYAML==6.0.2",
        "requests==2.32.3",
        "setuptools==49.2.1",
        "six==1.16.0",
        "texttable==1.6.3",
        "toml==0.10.2",
        "urllib3==1.26.5",
        "websocket-client==0.59.0"
    ],
    "usb": [
        "NoLib"
    ],
    "uname": [
        "Linux carter-calculon-20 5.15.0-131-generic #141~20.04.1-Ubuntu SMP Thu Jan 16 18:38:51 UTC 2025 x86_64 x86_64"
    ]
}
System info gathered successfully - saved as "log_system_information.json"
```

**Additional context**
Wireshark IO graph shows the network is healthy and sending data at roughly the expected rate.
[image]

If I drop the framerate from 20FPS to 2FPS:
```
[2821073132] [192.168.15.2] [1738712991.330] [Camera(0)] [debug] Camera timings: color_convert: 33ms
[2821073132] [192.168.15.2] [1738712991.453] [XLinkOut(4)] [debug] XLinkOut total took 498ms, processing 122ms, getting_frames 376ms, sending_frames 0ms
latency= 0:00:00.234389
[2821073132] [192.168.15.2] [1738712991.825] [Camera(0)] [debug] Camera timings: color_convert: 27ms
[2821073132] [192.168.15.2] [1738712991.948] [XLinkOut(4)] [debug] XLinkOut total took 493ms, processing 122ms, getting_frames 370ms, sending_frames 0ms
latency= 0:00:00.227305
```
I see 230ms latency... which is still way to high for my application.

Running at 2fps I can see with wireshark that transmission time is taking ~140ms per frame?
Which is fine considering 4000*3000*3 Bytes / 2.5 Gbps = 115ms. But even then where is the other 110ms of latency coming from?

[image]

For my application I have a need to access a full resolution RGB frame over the network with as low of latency as possible... If would be possible to send the raw frame prior to de-bayering that would substantially help latency here, but regardless it just seems like something is wrong?

Carter12s, 2025-02-05:
Note I tried to measure my link speed:
https://github.com/luxonis/depthai-experiments/blob/master/random-scripts/oak_bandwidth_test.py

But this script hasn't been updated for v3 yet.

Carter12s, 2025-02-05:
https://github.com/luxonis/depthai-experiments/blob/master/random-scripts/poe_test.py

Poe test also hasn't been updated yet as far as I can find

Carter12s, 2025-02-05:
Because: https://rvc4.docs.luxonis.com/software/depthai/optimizing/#Optimizing%20FPS%20%26%20Latency-Encoded%20frames

Indicated a 71ms "Time to Host" for a 4k video stream with MJPEG, I figured I should try a minimum example with VideoEncoder:

```Python
#!/usr/bin/env python3

import depthai as dai
import cv2

class VideoSaver(dai.node.HostNode):
    def __init__(self, *args, **kwargs):
        dai.node.HostNode.__init__(self, *args, **kwargs)

    def build(self, *args):
        self.link_args(*args)
        return self

    def process(self, frame: dai.EncodedFrame):
        print(f"Latency video: {dai.Clock.now() - frame.getTimestamp()}")

pipeline = dai.Pipeline()

# Seemed like it has a small effect 
# pipeline.setXLinkChunkSize(0)

cam = pipeline.create(dai.node.Camera).build()
output = cam.requestOutput((4000, 3000), dai.ImgFrame.Type.NV12, fps=2, enableUndistortion=False)
encoded = pipeline.create(dai.node.VideoEncoder).build(output, frameRate=2, profile=dai.VideoEncoderProperties.Profile.MJPEG)

saver = pipeline.create(VideoSaver).build(encoded.out)

pipeline.start()

while pipeline.isRunning():
    # Do nothing
    pass
    
pipeline.stop()
pipeline.wait()
```
This resulted in:
```
DEPTHAI_LEVEL=debug python3 ./carter_test.py 
[2025-02-04 17:29:52.608] [depthai] [debug] Python bindings - version: 3.0.0-alpha.12.dev0+19be0f8b89f39bef16fd81d47124c4ac15aad3d6 from  build: 2025-01-26 14:01:55 +0000
[2025-02-04 17:29:52.608] [depthai] [debug] Library information - version: 3.0.0-alpha.12, commit:  from , build: 2025-01-26 14:01:55 +0000, libusb enabled: true
[2025-02-04 17:29:52.618] [depthai] [debug] Initialize - finished
[2025-02-04 17:29:52.663] [depthai] [debug] Resources - Archive 'depthai-bootloader-fwp-0.0.26.tar.xz' open: 1ms, archive read: 53ms
[2025-02-04 17:29:52.914] [depthai] [debug] Resources - Archive 'depthai-device-fwp-976351fda3b70483e37882f989686bf0aa8c604d.tar.xz' open: 1ms, archive read: 304ms
[2025-02-04 17:29:53.154] [depthai] [info] Adding device to the filtered list: DeviceInfo(name=192.168.15.2, mxid=2821073132, X_LINK_GATE, X_LINK_TCP_IP, X_LINK_RVC4, X_LINK_SUCCESS)
[2025-02-04 17:29:53.154] [host] [debug] Device - OpenVINO version: universal
[2821073132] [192.168.15.2] [1738715393.154] [host] [debug] Device - BoardConfig: {"camera":[],"defaultImuExtr":{},"emmc":null,"gpio":[],"imu":null,"logDevicePrints":null,"logPath":null,"logSizeMax":null,"logVerbosity":null,"network":{"mtu":0,"xlinkTcpNoDelay":true},"nonExclusiveMode":false,"pcieInternalClock":null,"sysctl":[],"uart":[],"usb":{"flashBootedPid":63037,"flashBootedVid":999,"manufacturer":"","maxSpeed":4,"pid":63035,"productName":"","vid":999},"usb3PhyInternalClock":null,"uvc":null,"watchdogInitialDelayMs":null,"watchdogTimeoutMs":null} 
libnop:
0000: b9 13 b9 07 81 e7 03 81 3b f6 81 e7 03 81 3d f6 04 bd 00 bd 00 b9 02 00 01 ba 00 be be bb 00 bb
0020: 00 be be be be be be be 00 bb 00 be be bb 00
[2821073132] [192.168.15.2] [1738715393.346] [host] [debug] Schema dump: {"connections":[{"node1Id":1,"node1Output":"out","node1OutputGroup":"","node2Id":4,"node2Input":"in","node2InputGroup":""},{"node1Id":0,"node1Output":"0","node1OutputGroup":"dynamicOutputs","node2Id":1,"node2Input":"in","node2InputGroup":""}],"globalProperties":{"calibData":null,"cameraTuningBlobSize":null,"cameraTuningBlobUri":"","leonCssFrequencyHz":700000000.0,"leonMssFrequencyHz":700000000.0,"pipelineName":null,"pipelineVersion":null,"sippBufferSize":18432,"sippDmaBufferSize":16384,"xlinkChunkSize":-1},"nodes":[[4,{"alias":"","id":4,"ioInfo":[[["","in"],{"blocking":true,"group":"","id":5,"name":"in","queueSize":3,"type":3,"waitForMessage":false}]],"name":"XLinkOut","parentId":-1,"properties":[185,3,136,0,0,128,191,189,10,95,95,120,95,49,95,95,111,117,116,0]}],[1,{"alias":"","id":1,"ioInfo":[[["","out"],{"blocking":false,"group":"","id":4,"name":"out","queueSize":8,"type":0,"waitForMessage":false}],[["","bitstream"],{"blocking":false,"group":"","id":3,"name":"bitstream","queueSize":8,"type":0,"waitForMessage":false}],[["","in"],{"blocking":true,"group":"","id":2,"name":"in","queueSize":3,"type":3,"waitForMessage":false}]],"name":"VideoEncoder","parentId":-1,"properties":[185,11,0,30,0,0,0,0,4,80,0,0,136,0,0,0,64]}],[0,{"alias":"","id":0,"ioInfo":[[["dynamicOutputs","0"],{"blocking":false,"group":"dynamicOutputs","id":1,"name":"0","queueSize":8,"type":0,"waitForMessage":false}],[["","inputControl"],{"blocking":true,"group":"","id":0,"name":"inputControl","queueSize":3,"type":3,"waitForMessage":false}]],"name":"Camera","parentId":-1,"properties":[185,38,185,32,0,3,0,136,0,0,0,0,128,131,128,178,185,3,130,236,157,192,116,130,46,8,245,93,130,52,172,233,45,185,5,129,214,50,129,155,237,129,46,183,129,158,19,130,153,233,245,207,185,5,129,77,124,129,205,151,129,112,99,129,164,123,130,65,225,133,138,128,217,18,76,0,0,128,197,12,128,137,70,185,3,15,11,132,165,185,3,134,91,138,161,86,134,241,149,190,209,130,24,17,132,107,130,224,241,57,176,132,161,78,8,90,40,55,66,129,208,79,1,128,239,186,0,0,189,0,255,0,1,0,129,44,1,129,44,1,255,255,255,255,255,255,255,255,136,0,0,240,65,0,136,0,0,128,191,136,0,0,128,191,0,185,4,0,0,0,0,255,3,3,4,4,4,255,189,0,0,0,190,32,32,190,186,1,185,5,185,1,184,0,186,2,129,160,15,129,184,11,185,1,184,0,136,0,0,0,64,22,0,0]}]]}
[2821073132] [192.168.15.2] [1738715393.346] [host] [debug] Asset map dump: {"map":{}}
[2025-02-04 17:29:53.347] [depthai] [info] Pipeline not logged yet, logging...
[... cut ...]
[2025-02-04 17:30:00.330] [ThreadedNode] [debug] Starting input frame timestamp is 359694980 ms
[2025-02-04 17:30:00.330] [ThreadedNode] [debug] Diff: 0 ms
[2025-02-04 17:30:00.330] [ThreadedNode] [debug] Sync total took 500ms, processing 0ms, getting_frames 500ms, sending_frames 0ms
[2821073132] [192.168.15.2] [1738715391.951] [XLinkOut(4)] [debug] XLinkOut total took 501ms, processing 5ms, getting_frames 495ms, sending_frames 0ms
Latency video: 0:00:01.636838
```

And a measured 1.6 second time to host...

moratom, 2025-02-05:
Hi @Carter12s.

First of, super excited for getting the OAK 4 issues in, as that means people are starting to work with the cameras, so keep them coming!

As per the issue, you will need to lower the FPS to a point where the network is not a bottleneck (we recommend using NV12, since it only uses 1.5 bytes per pixel instead of 3 with BGR888i).

With 2.5 Gbps, you can get the maximum of 17 FPS with NV12 (2.5Gbps / (8 [bits in a byte] * 4000 [width] * 3000 [height] * 1.5[Bytes per pixel]) but more realistically, you can set 13 maybe 14 FPS to as there is some overhead.

For BGR888i (likely not the optimal choice here), you can get half that at the maximum, so ~7 FPS realistically.

moratom, 2025-02-05:
>However, measured latency for me is ~100ms. Indicating 43ms of processing time?
Actually according to logs `XLinkOut` takes 63 ms (since the actual throughput is likely a bit below the maximum of 2.5Gbps).
So only 27ms are spent processing the frame (frame needs to get from the sensor to the ISP and through the ISP pipeline).

We're just working on adding support for getting raw out directly, but I'm afraid we won't see drastic improvements.

Depending on the usecase though, you might benefit a lot with standalone mode, as you will be able to shave off those 63ms if you do the processing you need with raw frames on the camera directly.

Carter12s, 2025-02-05:
Gotcha! Yeah that is making more sense. Getting a better understanding as I work with the camera more.

To describe our use-case a bit, we have an extremely heavy object classification model that we run on sever grade GPUs. It is unfortunately orders of magnitude larger than what the Standalone Mode can handle. We're hoping to have onboard processing handle depth and object tracking with the classification occurring off-board.

We're currently evaluating this camera against a similar resolution unit from Basler that is able to get us a RAW frame over the network in right around 50ms (BUT, Basler unit supports a 5Gbps connection so cheating a bit...). Our application is extremely latency sensitive as these cameras directly control sorting equipment, and the 50ms difference is pretty meaningful to us.

Even if we could shave off 10-20ms by switching to RAW and 1 byte / pixel instead of 1.5 byte / pixel that would be very meaningful to us!

Please let me know when direct RAW capability is available, and I'd be happy to guinea pig and test early builds.

Also in general, THANK YOU for awesome support!

closed 2025-02-05 by Carter12s