luxonis/depthai-core #1235: Attempts to install `depthai` v3 fail with build errors seemingly caused by a cmake issue.

TheTridentGuy, 2025-02-07:
Since the [instructions](https://rvc4.docs.luxonis.com/software/depthai/examples/april_tags_12mp/) linked from [this issue](https://github.com/luxonis/depthai-python/issues/1096), don't specify what branch to checkout, and the default is still on depthai v2, I'm assuming its the `v3_develop` branch. There's a decent chance that I could be on the wrong branch, and if so please update the docs. When attempting to run `python3 install_requirements.py`, the following subcommand (shown isolated) drops an error that seems result from a cmake command failing. Any help is much appreciated :D

```
pi@luxonis:~/depthai-core $ /usr/bin/python3 -m pip install -U --prefer-binary --user --extra-index-url https://artifacts.luxonis.com/artifactory/luxonis-python-snapshot-local "depthai==3.0.0-alpha.12.dev0+4ca505bed06df4c87d7d8e4e23d7527e64d203e3"
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple, https://artifacts.luxonis.com/artifactory/luxonis-python-snapshot-local
Collecting depthai==3.0.0-alpha.12.dev0+4ca505bed06df4c87d7d8e4e23d7527e64d203e3
  Using cached https://artifacts.luxonis.com/artifactory/luxonis-python-snapshot-local/depthai/depthai-3.0.0a12.dev0%2B4ca505bed06df4c87d7d8e4e23d7527e64d203e3.tar.gz (213 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Building wheels for collected packages: depthai
  Building wheel for depthai (pyproject.toml) ... error
  error: subprocess-exited-with-error
  
  × Building wheel for depthai (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [68 lines of output]
      /tmp/pip-build-env-j3xprqpt/overlay/lib/python3.9/site-packages/setuptools/dist.py:330: InformationOnly: Normalizing '3.0.0-alpha.12.dev0+4ca505bed06df4c87d7d8e4e23d7527e64d203e3' to '3.0.0a12.dev0+4ca505bed06df4c87d7d8e4e23d7527e64d203e3'
        self.metadata.version = self._normalize_version(self.metadata.version)
      running bdist_wheel
      running build
      running build_py
      creating build/lib.linux-armv7l-cpython-39/depthai_cli
      copying depthai_cli/depthai_cli.py -> build/lib.linux-armv7l-cpython-39/depthai_cli
      copying depthai_cli/__init__.py -> build/lib.linux-armv7l-cpython-39/depthai_cli
      running build_ext
      CMake Warning:
        Ignoring extra path from command line:
      
         "/tmp"
      
      
      CMake Error: The source directory "/tmp" does not appear to contain CMakeLists.txt.
      Specify --help for usage, or press the help button on the CMake GUI.
      Traceback (most recent call last):
        File "/home/pi/.local/lib/python3.9/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 389, in <module>
          main()
        File "/home/pi/.local/lib/python3.9/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 373, in main
          json_out["return_val"] = hook(**hook_input["kwargs"])
        File "/home/pi/.local/lib/python3.9/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 280, in build_wheel
          return _build_backend().build_wheel(
        File "/tmp/pip-build-env-j3xprqpt/overlay/lib/python3.9/site-packages/setuptools/build_meta.py", line 438, in build_wheel
          return _build(['bdist_wheel', '--dist-info-dir', str(metadata_directory)])
        File "/tmp/pip-build-env-j3xprqpt/overlay/lib/python3.9/site-packages/setuptools/build_meta.py", line 426, in _build
          return self._build_with_temp_dir(
        File "/tmp/pip-build-env-j3xprqpt/overlay/lib/python3.9/site-packages/setuptools/build_meta.py", line 407, in _build_with_temp_dir
          self.run_setup()
        File "/tmp/pip-build-env-j3xprqpt/overlay/lib/python3.9/site-packages/setuptools/build_meta.py", line 522, in run_setup
          super().run_setup(setup_script=setup_script)
        File "/tmp/pip-build-env-j3xprqpt/overlay/lib/python3.9/site-packages/setuptools/build_meta.py", line 320, in run_setup
          exec(code, locals())
        File "<string>", line 220, in <module>
        File "/tmp/pip-build-env-j3xprqpt/overlay/lib/python3.9/site-packages/setuptools/__init__.py", line 117, in setup
          return distutils.core.setup(**attrs)
        File "/tmp/pip-build-env-j3xprqpt/overlay/lib/python3.9/site-packages/setuptools/_distutils/core.py", line 186, in setup
          return run_commands(dist)
        File "/tmp/pip-build-env-j3xprqpt/overlay/lib/python3.9/site-packages/setuptools/_distutils/core.py", line 202, in run_commands
          dist.run_commands()
        File "/tmp/pip-build-env-j3xprqpt/overlay/lib/python3.9/site-packages/setuptools/_distutils/dist.py", line 983, in run_commands
          self.run_command(cmd)
        File "/tmp/pip-build-env-j3xprqpt/overlay/lib/python3.9/site-packages/setuptools/dist.py", line 999, in run_command
          super().run_command(command)
        File "/tmp/pip-build-env-j3xprqpt/overlay/lib/python3.9/site-packages/setuptools/_distutils/dist.py", line 1002, in run_command
          cmd_obj.run()
        File "/tmp/pip-build-env-j3xprqpt/overlay/lib/python3.9/site-packages/setuptools/command/bdist_wheel.py", line 379, in run
          self.run_command("build")
        File "/tmp/pip-build-env-j3xprqpt/overlay/lib/python3.9/site-packages/setuptools/_distutils/cmd.py", line 339, in run_command
          self.distribution.run_command(command)
        File "/tmp/pip-build-env-j3xprqpt/overlay/lib/python3.9/site-packages/setuptools/dist.py", line 999, in run_command
          super().run_command(command)
        File "/tmp/pip-build-env-j3xprqpt/overlay/lib/python3.9/site-packages/setuptools/_distutils/dist.py", line 1002, in run_command
          cmd_obj.run()
        File "/tmp/pip-build-env-j3xprqpt/overlay/lib/python3.9/site-packages/setuptools/_distutils/command/build.py", line 136, in run
          self.run_command(cmd_name)
        File "/tmp/pip-build-env-j3xprqpt/overlay/lib/python3.9/site-packages/setuptools/_distutils/cmd.py", line 339, in run_command
          self.distribution.run_command(command)
        File "/tmp/pip-build-env-j3xprqpt/overlay/lib/python3.9/site-packages/setuptools/dist.py", line 999, in run_command
          super().run_command(command)
        File "/tmp/pip-build-env-j3xprqpt/overlay/lib/python3.9/site-packages/setuptools/_distutils/dist.py", line 1002, in run_command
          cmd_obj.run()
        File "<string>", line 93, in run
        File "<string>", line 217, in build_extension
        File "/usr/lib/python3.9/subprocess.py", line 373, in check_call
          raise CalledProcessError(retcode, cmd)
      subprocess.CalledProcessError: Command '['cmake', '/tmp', '-DDEPTHAI_BUILD_PYTHON=ON', '-DCMAKE_LIBRARY_OUTPUT_DIRECTORY=/tmp/pip-install-xw7y7a9f/depthai_fb2cef16393f47cd9316debf0539fb41/build/lib.linux-armv7l-cpython-39/', '-DPYTHON_EXECUTABLE=/usr/bin/python3', '-DDEPTHAI_PYTHON_COMMIT_HASH=4ca505bed06df4c87d7d8e4e23d7527e64d203e3', '-DCMAKE_BUILD_TYPE=Release', '-DHUNTER_CONFIGURATION_TYPES=Release', '-DHUNTER_JOBS_NUMBER=1']' returned non-zero exit status 1.
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for depthai
Failed to build depthai
ERROR: Failed to build installable wheels for some pyproject.toml based projects (depthai)
```
System: Rapsberry Pi 4B 4GB running the luxonis image with an OAK-D camera connected.

moratom, 2025-02-07:
Hi @TheTridentGuy !

Thanks for the issue report.

We don't yet support 32bit wheels, we recommend using 64bit Raspberry OS if you can in the mean time.

TheTridentGuy, 2025-02-07:
OK, is there anything I should keep in mind when setting up a 64-bit system, i.e. quirks, etc?

moratom, 2025-02-07:
Nothing specific, it's well supported for Raspberry OS, just make sure to select the 64bit version and you should be good to go.

TheTridentGuy, 2025-02-07:
Alright, thanks for the help, I'll go ahead and close the issue.

closed 2025-02-07 by TheTridentGuy