luxonis/depthai-core #1239: [BUG] IMX582 (Oak-1 Max) out of memory for still capture at 5312X6000

timlod, 2025-02-12:
**Describe the bug**
When setting up the Oak-1 Max for still capture at max resolution, we get the following error:
`RuntimeError: ColorCamera(0) - Out of memory while creating pool for 'still' frames. Number of frames: 4 each with size: 47808000B`

**Minimal Reproducible Example**
```python
pipeline = dai.Pipeline()
cam_rgb = pipeline.createColorCamera()
cam_rgb.setResolution(dai.ColorCameraProperties.SensorResolution.THE_5312X6000)
xout_still = pipeline.createXLinkOut()
xout_still.setStreamName("still")
cam_rgb.still.link(xout_still.input)
device = dai.Device(pipeline)
```

```python
---------------------------------------------------------------------------
RuntimeError                              Traceback (most recent call last)
Cell In[242], line 1
----> 1 device = dai.Device(pipeline)

RuntimeError: ColorCamera(0) - Out of memory while creating pool for 'still' frames. Number of frames: 4 each with size: 47808000B
```

**Expected behavior**
It should work fine.

**Attach system log**
```
{
    "architecture": "64bit ELF",
    "machine": "x86_64",
    "platform": "Linux-6.9.3-76060903-generic-x86_64-with-glibc2.35",
    "processor": "x86_64",
    "python_build": "main Dec  4 2024 08:55:07",
    "python_compiler": "GCC 11.4.0",
    "python_implementation": "CPython",
    "python_version": "3.11.11",
    "release": "6.9.3-76060903-generic",
    "system": "Linux",
    "version": "#202405300957~1738770968~22.04~d5f7c84 SMP PREEMPT_DYNAMIC Wed F",
    "win32_ver": "",
    "packages": [
        "blobconverter==1.4.3",
        "certifi==2024.12.14",
        "charset-normalizer==2.0.12",
        "Deprecated==1.2.15",
        "depthai==2.24.0.0",
        "depthai-sdk==1.9.4",
        "depthai-viewer==0.2.8",
        "distinctipy==1.3.4",
        "ffmpy3==0.2.4",
        "idna==3.10",
        "lz4==4.3.3",
        "marshmallow==3.17.0",
        "mcap==1.2.2",
        "mcap-ros1-support==0.0.8",
        "numpy==1.26.4",
        "opencv-contrib-python==4.5.5.62",
        "packaging==24.2",
        "pip==24.3.1",
        "pyarrow==16.0.0",
        "PyQt5==5.15.5",
        "PyQt5-Qt5==5.15.16",
        "PyQt5_sip==12.16.1",
        "pytube==15.0.0",
        "PyTurboJPEG==1.6.4",
        "pyusb==1.2.1",
        "PyYAML==6.0.2",
        "Qt.py==1.4.1",
        "requests==2.26.0",
        "rosbags==0.9.11",
        "ruamel.yaml==0.18.10",
        "ruamel.yaml.clib==0.2.12",
        "scipy==1.15.1",
        "sentry-sdk==1.14.0",
        "setuptools==75.8.0",
        "types-PySide2==5.15.2.1.7",
        "urllib3==1.26.20",
        "wrapt==1.17.2",
        "xmltodict==0.14.2",
        "zstandard==0.23.0"
    ],
    "usb": [
        {
            "port": 0,
            "vendor_id": "0x1d6b",
            "product_id": "0x0002",
            "speed": "High"
        },
        {
            "port": 0,
            "vendor_id": "0x1d6b",
            "product_id": "0x0003",
            "speed": "SuperPlus"
        },
        {
            "port": 1,
            "vendor_id": "0x2109",
            "product_id": "0x0817",
            "speed": "Super"
        },
        {
            "port": 0,
            "vendor_id": "0x1d6b",
            "product_id": "0x0003",
            "speed": "SuperPlus"
        },
        {
            "port": 5,
            "vendor_id": "0x2109",
            "product_id": "0x8817",
            "speed": "High"
        },
        {
            "port": 4,
            "vendor_id": "0x03e7",
            "product_id": "0x2485",
            "speed": "High"
        },
        {
            "port": 3,
            "vendor_id": "0x2109",
            "product_id": "0x0103",
            "speed": "Full"
        },
        {
            "port": 2,
            "vendor_id": "0x0bda",
            "product_id": "0x8153",
            "speed": "High"
        },
        {
            "port": 1,
            "vendor_id": "0x1209",
            "product_id": "0x2201",
            "speed": "Full"
        },
        {
            "port": 1,
            "vendor_id": "0x2109",
            "product_id": "0x2817",
            "speed": "High"
        },
        {
            "port": 0,
            "vendor_id": "0x1d6b",
            "product_id": "0x0002",
            "speed": "High"
        },
        {
            "port": 0,
            "vendor_id": "0x1d6b",
            "product_id": "0x0003",
            "speed": "Super"
        },
        {
            "port": 1,
            "vendor_id": "0x174f",
            "product_id": "0x1812",
            "speed": "High"
        },
        {
            "port": 0,
            "vendor_id": "0x1d6b",
            "product_id": "0x0002",
            "speed": "High"
        },
        {
            "port": 1,
            "vendor_id": "0x04e8",
            "product_id": "0x4001",
            "speed": "Super"
        },
        {
            "port": 0,
            "vendor_id": "0x1d6b",
            "product_id": "0x0003",
            "speed": "SuperPlus"
        },
        {
            "port": 3,
            "vendor_id": "0x06cb",
            "product_id": "0x00f9",
            "speed": "Full"
        },
        {
            "port": 0,
            "vendor_id": "0x1d6b",
            "product_id": "0x0002",
            "speed": "High"
        },
        {
            "port": 0,
            "vendor_id": "0x1d6b",
            "product_id": "0x0003",
            "speed": "SuperPlus"
        },
        {
            "port": 1,
            "vendor_id": "0x10ab",
            "product_id": "0x9309",
            "speed": "Full"
        },
        {
            "port": 3,
            "vendor_id": "0x05e3",
            "product_id": "0x0610",
            "speed": "High"
        },
        {
            "port": 0,
            "vendor_id": "0x1d6b",
            "product_id": "0x0002",
            "speed": "High"
        }
    ],
    "uname": [
        "Linux popmachine 6.9.3-76060903-generic #202405300957~1738770968~22.04~d5f7c84 SMP PREEMPT_DYNAMIC Wed F x86_64 x86_64"
    ]
}
```

timlod, 2025-02-13:
I noticed that it will work with the video output unless a control is linked, then we get the same error.
`isp` works with both max resolution and control linked. 
I'm not sure why control changes RAM, but I found this:
https://docs.luxonis.com/hardware/sensors/IMX582#IMX582-RVC2%20driver%20limits-IMX582%20RAM%20consumption
which suggests that we need to change pool sizes for things to work.
It might be helpful to add this to the RAM error message, or if a resolution could define the pools in such a way that it works - in the case of the IMX582, I imagine the reason people get this sensor is to shoot with those resolutions. Ideally it would work 'out-of-the-box' without custom settings.

moratom, 2025-02-18:
Thanks for the report @timlod !

We'll look into downsizing the pool sizes automatically for big resolutions, it will technically be a breaking change, since we now have well defined defaults, but it's worth doing, if not elsewhere than on the next major version of the library.