luxonis/depthai-core #1252: [BUG] CMake Integration Issues on v3.0.0-alpha12

Carter12s, 2025-02-21:
**Describe the bug**
It appears that certain 3rd party dependencies (notably nlohmann json) that are provided by vcpkg are not incorporated correctly into the build / install.

This causes builds to fail if following the README "Integration" section instructions.

**Minimal Reproducible Example**
I'm attempting to setup a minimal docker image for compiling with this library (#1228). It is currently as follows:

```
FROM ubuntu:20.04
ARG DEBIAN_FRONTEND=noninteractive
# Install base packages
RUN apt-get update && apt-get install -y \
    wget \
    curl \
    zip \
    unzip \
    tar \
    build-essential \
    cmake \
    pkg-config \
    libjpeg-dev \
    libtiff5-dev \
    libavcodec-dev \
    libavformat-dev \
    libswscale-dev \
    libv4l-dev \
    libxvidcore-dev \
    libx264-dev \
    libgtk2.0-dev \
    libgtk-3-dev \
    libatlas-base-dev \
    gfortran \
    git

# This was adapted from: https://github.com/luxonis/depthai/blob/main/docker_dependencies.sh
RUN apt-get update && apt-get install -y \
    # Yes python is a required part of this build system :facepalm:
    python3 \
    python3-pip \
    python3-numpy \
    udev \
    libudev-dev \
    libgtk2.0-dev \
    libtbb2 \
    libtbb-dev \
    libjpeg-dev \
    libpng-dev \
    libtiff-dev \
    libdc1394-22-dev \
    ffmpeg \
    libsm6 \
    libxext6 \
    libgl1-mesa-glx \
    bison \
    nasm \
    libarchive-dev

# No idea why this is needed, but all over luxonis documentation
RUN python3 -m pip install --upgrade pip
# Random missing python dependency needed by build system
RUN pip install jinja2

# Clone in the workspace
WORKDIR /workspaces
RUN git clone --depth 1 --branch v3.0.0-alpha.12 https://github.com/luxonis/depthai-core.git
WORKDIR /workspaces/depthai-core
RUN git submodule update --init --recursive

# Doing a really crappy find and replace to fix a bug in this version of the library
RUN sed -i 's/LibArchive::LibArchive/archive/g' CMakeLists.txt

# Build and install with shared library
# RUN cmake -S . -B build -D CMAKE_BUILD_TYPE=Release -D BUILD_SHARED_LIBS=ON -D CMAKE_INSTALL_PREFIX=/usr/local
RUN cmake -S . -B build -D CMAKE_BUILD_TYPE=Release -D BUILD_SHARED_LIBS=ON
# RUN cmake --build build --parallel 8 --target install
RUN cmake --build build --parallel 8
```

After generating this docker image it can successfully build and install the package.

However, attempting to then use the package reveals that not all dependencies were installed.

Setting up a basic CMake project and attempting to manually setup header / linking will reveal that nlomann/json.hpp is missing.

Using `find_package(depthai)` in CMake will reveal that `nlohmann_jsonConfig.cmake` is missing.

It is possible that the "add_subdirectory" version of integration does work, I'm running into separate problems with that.

**Expected behavior**
Following the first two options in the "Integration" section of README.md should work.

moratom, 2025-02-24:
Hi @Carter12s - thanks for the issue report!

We haven't had a chance to provide proper integration support as a C++ package yet for v3 since we switched to `vcpkg`.

We'll still focusing our effort on RVC4 functionality and we plan to add proper integration in the second half of March 2025.

Carter12s, 2025-02-24:
Ahh that makes sense that it is in flight!

No worries! Thanks for all the awesome work on this, and I'll keep an eye as things get updated.

Will close this issue for now as probably noise until this is supported.

closed 2025-02-24 by Carter12s