luxonis/depthai-core #1289: [BUG] Incorrect Still Image Resolution -  OAK-FFC IMX577 M12 -  12 MP

Krokoman13, 2025-04-01:
A returned still image from the OAK-FFC IMX577 M12 set to SensorResolution::THE_12_MP returns a frame of 4032 x 3040 (expected resolution: 4056x3040).

Minimal reproducible example:
Connect OAK-FFC IMX577 M12  to my PC,
```
dai::Pipeline pipeline;

 auto camRgb = pipeline.create<dai::node::ColorCamera>();
 camRgb->setBoardSocket(dai::CameraBoardSocket::CAM_A);
 camRgb->setResolution(dai::ColorCameraProperties::SensorResolution::THE_12_MP);

 auto stillOut = pipeline.create<dai::node::XLinkOut>();
 stillOut->setStreamName("still");
 camRgb->still.link(stillOut->input);

 auto controlIn = pipeline.create<dai::node::XLinkIn>();
 controlIn->setStreamName("control");
 controlIn->out.link(camRgb->inputControl);

 dai::Device device(pipeline);
 auto stillQ = device.getOutputQueue("still", 1, true);
 auto controlQ = device.getInputQueue("control");

 dai::CameraControl ctrl;
 ctrl.setCaptureStill(true);
 controlQ->send(ctrl);

 auto imgFrame = stillQ->get<dai::ImgFrame>();

 cv::Mat cvFrame = imgFrame->getCvFrame();
 std::cout << "Captured frame resolution: "
     << cvFrame.cols << " x " << cvFrame.rows << std::endl;
```

Expected resolution: 4056x3040
Actual resolution: 4032 x 3040 

As far as we know, setting the resolution of the camera should also set the resolution of the still image.
When testing we did realize this problem does NOT occur when using the ISP queue, it does return an correct image.
This problem persists both on Python and C++

[log_system_information.json](https://github.com/user-attachments/files/19551776/log_system_information.json)

moratom, 2025-04-01:
@alex-luxonis would you maybe know what's the cause here?

alex-luxonis, 2025-04-01:
To get the full resolution, add: `camRgb->setStillSize(4056, 3040);`

The default of 4032 here is due to historical reasons, usually the still output was connected to a JPEG encoder, and the encoder needs the width to be multiple of 64.

Krokoman13, 2025-04-02:
We did know of that solution.

The reason is interesting...
But is it considered intended behaviour that setResolution seems to also set the stillSize for every other resolution?
Or that after setting the resolution that the ISP resolution is different to the still resolution?

Krokoman13, 2025-04-04:
@alex-luxonis so as far as I understand you don't consider this a bug and we should close it?