luxonis/depthai-core #1307: missing include in file

MaximBozek-M-ECS, 2025-04-23:
I am getting this error for the include of the depthai library:

```
main.cpp:9:10: In included file: 'nop/serializer.h' file not found
Serialization.hpp:9:10: error occurred here
```

any idea what I'm missing?

MaximBozek-M-ECS, 2025-04-24:
Is it normal that I had to install an additional 5 libraries to get this (not even) working? Is this library usable?

now getting this error:

```
main.cpp:9:10: In included file: no member named 'Empty' in 'nop::EncodingByte'; did you mean 'QtPrivate::QContainerImplHelper::Empty'?
optional.hpp:57:54: error occurred here
qarraydata.h:350:28: 'QtPrivate::QContainerImplHelper::Empty' declared here
```

MaximBozek-M-ECS, 2025-04-25:
I git cloned the HTTPS [link](https://github.com/luxonis/depthai-core.git)

I then followed the building instructions, these are all commands I carried out:

```
git submodule update --init --recursive
cmake -S. -Bbuild -D'BUILD_SHARED_LIBS=ON'
cmake --build build
```

With this, I still had to manually add: libnop, nlohmann/json, XLink and optional. After doing this, I still had errors like:

```
main.cpp:9:10: In included file: no member named 'Empty' in 'nop::EncodingByte'; did you mean 'QtPrivate::QContainerImplHelper::Empty'?
optional.hpp:57:54: error occurred here
qarraydata.h:350:28: 'QtPrivate::QContainerImplHelper::Empty' declared here
```

moratom, 2025-05-05:
@MaximBozek-M-ECS this is not expected at all, can you share the whole logs or ideally an example where this happen that is dockerized so our team can take a look and help you out?

We fetch all the dependanices with Hunter, so these should be resolved by our own library.

moratom, 2025-05-05:
From the logs, it seems like you're missing our public dependanices headers in your `main.cpp` file when compiling.

You can see this for an example on how to use library with adding `add_subdirectory` https://github.com/luxonis/depthai-core-example or install the library with `DEPTHAI_INSTALL=ON` during configure time.