luxonis/depthai-core #1320: [BUG] depthai-core does not build on Mac OSX Sequoia 15.4.1

dizzyd, 2025-05-16:
**Describe the bug**

A fresh checkout of depthai-core fails to build on OSX 15.4.1, failing (initially) when attempting to build zlib. I believe that this is because the version of zlib used does not compile when using a C23 compiler (which is the default compiler on this version of OSX). 

Once this issue is resolved, the next failure is nholmann/json - it's doing some questionable template stuff that the compiler hard errors on (and you can't force the error to a warning):

.hunter/_Base/cb0ea1f/8c87124/7ad1efd/Install/include/nlohmann/json.hpp:5989:41: error: implicit instantiation of undefined template 'std::char_traits<unsigned char>'

Bottom line, I don't believe anyone has tried to compile this on a recent version of OSX. :) 

**Minimal Reproducible Example**

1. Checkout depthai-core repo on OSX 15.4.1
2. Attempt to compile following instructions in README.md
3. See zlib error
4. If you somehow work around zlib error (I used a custom FindZLib cmake macro), compilation will then fail in nholmann/json.

**Expected behavior**

depthai-core should compile cleanly on OSX.

marc-medley, 2025-05-20:
@dizzyd 

> Once this issue [build zlib] is resolved, the next failure is …

What approach(es) did you use to resolve the build zlib stage of this issue?

One underlaying zlib problem is that the [luxinos/zlib fork v1.2.11-p2](https://github.com/luxonis/zlib) version has a `#if defined(MACOS) || defined(TARGET_OS_MAC)` … `#define fdopen(fd,mode) NULL` statement. 

The newer macOS build chain now defines `fdopen`, so any (not-patched-for-this-issue) zlib prior to 1.3.1 will have current macOS tools build issues.

See related: "Update ZLIB to v1.3.1" https://github.com/cpp-pm/hunter/issues/735#issuecomment-2892485545

dizzyd, 2025-05-20:
@marc-medley - as noted in the bug report, I used a custom FindZLib macro that picked up zlib 1.3.1 from homebrew for exactly the reasons you stated re: fdopen.

If that had been the only issue, I would have been happy with it - but finding that there were also weird template issues with the (quite old) nholmann/json library, I decided that this is probably an issue for maintainers to work through. :) There are a number of unusual forks that Luxonis has picked up as part of the depthai-core dependency tree (fp16, libnop, zlib, etc) that probably need updating.

moratom, 2025-05-21:
Hi all, thanks for reporting!

We've been shifting a lot of focus to the `v3.x.y` version of the library, which we'll be release a beta version of this week.

For the issues above, I wonder if it's possible by pinning down C++ version to C++14/17 in the toolchain file when compiling the library?

marc-medley, 2025-05-26:
@dizzyd As fyi, I did manage to get a successful build on macOS 15.4.1 from the `v3_develop` branch. See [[BUG/v3_develop] macOS Sequoia 15.4.1 depthai-core build failure (w/some resolution approaches) #1333](https://github.com/luxonis/depthai-core/issues/1333) for details.

matlabbe, 2025-07-13:
@dizzyd Following your [first fix](https://github.com/luxonis/depthai-core/issues/1320#issuecomment-2895681201) for zlib (I replaced the `findZLIB.cmake` used by Hunter by this [one](https://github.com/Kitware/VTK/blob/527800fd1820ee6facfc831f45658f16915f729b/CMake/patches/3.13/FindZLIB.cmake#L4)), the only thing remaining to fix the undefined `std::char_traits` error is to replace the whole `json.hpp` file linked in the error (`.hunter/_Base/cb0ea1f/8c87124/7ad1efd/Install/include/nlohmann/json.hpp`in your case) by the latest version here https://github.com/nlohmann/json/tree/develop/single_include/nlohmann