luxonis/depthai-core #1324: [BUG] build fails with current CMake 4.0.2

marc-medley, 2025-05-18:
**Describe the bug**

> "Compatibility with CMake < 3.5 has been removed from CMake"

The files "depthai-core/CMakeLists.txt" and "depthai-core/tests/integration/CMakeLists.txt" have the statement `cmake_minimum_required(VERSION 3.4)` which is no longer supported with CMake 4+.

- [depthai-core/CMakeLists.txt](https://github.com/luxonis/depthai-core/blob/main/CMakeLists.txt)
- [depthai-core/tests/integration/CMakeLists.txt](https://github.com/luxonis/depthai-core/blob/main/tests/integration/CMakeLists.txt)

**Minimal Reproducible Example**

Use the [depthai-core README.md "Building" instructions](https://github.com/luxonis/depthai-core?tab=readme-ov-file#building) on a clean download (or clone) of this repository.

The following CMake Error occurs:

```sh
# Configuring build...
# CMake Error at CMakeLists.txt:1 (cmake_minimum_required):
#   Compatibility with CMake < 3.5 has been removed from CMake.
```

**Expected behavior**

Successful build of depthai-core with the currently released cmake which is [version 4.0.2](https://cmake.org/download/).

**Attach system log**

This issue applies across all systems which use the current release of CMake 4.0.2,  which no longer supports CMake prior to 3.6.

moratom, 2025-05-19:
Hi @marc-medley - thanks for bringing it up.

Due to quite a large dependancy tree, where now many of the dependanices don't configure with CMake 4.x.y, we'll need some time to address.
In the meantime, please consider downgrading CMake for DepthAI to 3.x.y or you can try using `-DCMAKE_POLICY_VERSION_MINIMUM=3.5`, just note that this wasn't tested internally.

marc-medley, 2025-05-20:
@moratom Thanks for the reply.

> … consider downgrading CMake …

Downgrading is not really an option for some the systems that I work with. Here's one workaround that I'm using:

```sh
export CMAKE3="/path/to/3.31.7/bin/cmake"
export CMAKE4="/path/to/4.0.2/bin/cmake"

# … elsewhere …
"$CMAKE3" -S .. -B .
```

> … or you can try using `-DCMAKE_POLICY_VERSION_MINIMUM=3.5`, just note that this wasn't tested internally.

As fyi, limited testing here of setting `CMAKE_POLICY_VERSION_MINIMUM` has not resolved the cmake version issues for `depthai-core`.

marc-medley, 2025-05-22:
Note: Homebrew does not currently have a `cmake@3` formula to provide an alternate to the cmake v4.0.2 now available via homebrew. 

So, an approach is to install the [CMake.app binary dmg downloads](https://cmake.org/download/) as `CMake_3.31.7.app` and `CMake_4.0.2.app`

These CMake apps can then be setup to use cmake v3 and cmake v4 from the command line or scripts … 

```
# mkdir /opt/cmake/bin
cd /opt/cmake/bin 
LINK_NAME='cmake3'
SOURCE_URL="/Applications/CMake_3.31.7.app/Contents/bin/cmake"
ln -sFf "$SOURCE_URL" "$LINK_NAME"

LINK_NAME='cmake4'
SOURCE_URL="/Applications/CMake_4.0.2.app/Contents/bin/cmake"
ln -sFf "$SOURCE_URL" "$LINK_NAME"
```

_.zshrc_

```sh
#############
### CMAKE ###
#############
### To use /opt/cmake/bin/cmake3*
export PATH="$PATH:/opt/cmake/bin"
### To use `$CMAKE3`
export CMAKE3="/Applications/CMake_3.31.7.app/Contents/bin/cmake"
export CMAKE4="/Applications/CMake_4.0.2.app/Contents/bin/cmake"
export CMAKE="$CMAKE3"
```

Caveat: 

- `/opt/cmake/bin/cmake` was not set up because `cmake` might conflict with what Homebrew expects at `/opt/homebrew/bin/cmake`
- `/opt/cmake/bin/` could be some other location of choice.

marc-medley, 2025-05-26:
@moratom 

> Due to quite a large dependancy tree ...

Unfortunately, the v2 depedency tree has more issues that just being "large". The depedency tree has a range of various patches (including on [critically outdated versions](https://github.com/luxonis/depthai-core/issues/1328)), and the Hunter dependency repository itself is not up-to-date with upstream releases. For example, [cpp-pm/hunter Issue: Update ZLIB to v1.3.1 #735](https://github.com/cpp-pm/hunter/issues/735)

That said, I was successful in building `v3_develop` with `cmake` v4.0.2. See cmake note at bottom of [[BUG/v3_develop] macOS Sequoia 15.4.1 depthai-core build failure (w/some resolution approaches) #1333](https://github.com/luxonis/depthai-core/issues/1333).

So, `depthai-core` v3 looks promising for use with cmake 4.