luxonis/depthai-core #1333: [BUG/v3_develop] macOS Sequoia 15.4.1 depthai-core build failure (w/some resolution approaches)

marc-medley, 2025-05-26:
**Describe Issue**

A cmake `v3_develop` branch snapshot was found to build OK on a Raspberry Pi OS (Debian) RPi5. However, the same snapshot failed to build with a current macOS 15.4.1 / Xcode 16.3 / Apple clang version 17.0.0 (clang-1700.0.13.3) toolchain.

Steps were then taken to achieve a successful macOS 15.4.1 build. The steps for a successful build are outlined in the **Additional Context** section. Also, the successfully built snapshot is provided on the `v3_develop_macos15` branch of the `marc-medley/depthai-core` fork. 

- [marc-medley/depthai-core/v3_develop_macos15](https://github.com/marc-medley/depthai-core/tree/v3_develop_macos15)

**Minimal Reproducible Example**

Use a macOS with an up-to-date toolchain. In this case, a macOS 15.4.1 / Xcode 16.3 / Apple clang version 17.0.0 (clang-1700.0.13.3) toolchain was used.

```sh
brew install opencv pcl xtensor clang-format

git clone --single-branch \
    --branch v3_develop \
    --recurse-submodules \
    git@github.com:luxonis/depthai-core.git \
    depthai_v3_develop
cd depthai_v3_develop

mkdir build && cd build
cmake -S .. -B . \
    -D'CMAKE_POLICY_VERSION_MINIMUM=3.5'

NUM_CORES=$(sysctl -n hw.ncpu) # optional
cmake --build .  --parallel $NUM_CORES
```

_Result:_

- `xtensor`, `xtl` failures blocked the build
- some cpp examples failed to build
- 1,628 warnings due to a SDK major version mismatch
- ClangFormat warnings

**Expected behavior**

A successful `depthai-core` v3.x.y-beta-like build on an up-to-date macOS toolchain.

**Attach System Log**

[log_system_information.json](https://github.com/user-attachments/files/20435956/log_system_information.json)

**Additional Context**

_xtensor, xtl_

The code build issues within `xtensor` v0.25.0, `xtl` v0.7.6 were resolved by updating to the `xtensor` v0.26.0, `xtl` v0.8.0 releases.

- [Commit 696edfd: macOS 15 build - xtensor 0.26.0, xtl 0.8.0](https://github.com/marc-medley/depthai-core/commit/696edfd50bf0d658b7bb365bcae96ad497098425)

_Rerun Examples_

Two cpp examples (RGBD, RVC2/VSLAM) failed to build because the Apache Arrow depenency of [rerun-io/rerun 0.16.1](https://github.com/rerun-io/rerun/releases/tag/0.16.1) [2024.05.29] the failed to build.

A quick check of using the current release [rerun-io/rerun 0.23.2](https://github.com/rerun-io/rerun/releases/tag/0.23.2) [2025.05.06] also did not build. So, these two examples were just commented out instead of putting more effort into the examples at this time.

- [Commit 403a8ff: macOS 15 build - commented out VSLAM, RGBD examples due to rerun_cpp 0.16.1 build failure](https://github.com/marc-medley/depthai-core/commit/403a8ff7d88cb92bf7edb2cdb7d77748abfc9b2d)

_MACOS\_SDK\_MAJOR\_VERSION_

There were 1,628 ld warnings related to a SDK major version mismatch.

```
ld: warning: object file (…/build/_deps/….cpp.o)) 
    was built for newer 'macOS' version (15.0) than being linked (11.0)
```

The warnings were resolved by adding some automatic `MACOS_SDK_MAJOR_VERSION` detection.

- [Commit 7be7f07: macOS 15 build - adds automatic MACOS_SDK_MAJOR_VERSION detection ](https://github.com/marc-medley/depthai-core/commit/7be7f073b647449b8a99471c19bed91b031aec67)

_ClangFormat_

There were about 70 ClangFormat not available messages which were resolved with `brew install clang-format`.

```
-- ClangFormat: clang-format not found! Target 'clangformat' not available...
```

> Note: It would be helpful to have `brew install clang-format` added to the [depthai-core/v3_develop/README.md](https://github.com/luxonis/depthai-core/blob/v3_develop/README.md)

_CMake Versions_

- cmake version 3.31.7 also built OK without requiring `-D'CMAKE_POLICY_VERSION_MINIMUM=3.5'`.
- cmake 4.0.2 successfully builds the [marc-medley/depthai-core/v3_develop_macos15](https://github.com/marc-medley/depthai-core/tree/v3_develop_macos15) snapshot provided `-D'CMAKE_POLICY_VERSION_MINIMUM=3.5'` is used.

moratom, 2025-06-10:
@marc-medley can you give `v3.0.0-rc.2` a try? I think all of these should be addressed, except for the `MACOS_SDK_MAJOR_VERSION`, but we'll have to address that one a tad differently, to actually target all of the dependanices to the older MacOS, so the binaries are transferable.

marc-medley, 2025-09-13:
@moratom Regarding `xtensor`, 

> … did you explicitly change {this} to install xtensor via brew instead?

Here here are my steps and what was/was-not explicitly changed.

1. **Brew Updates.** As a background context, I routinely keep brew fully up-to-date. `brew update; brew upgrade` 

2. **`depthai-core` Standard Builds.** The `v3.0.0-rc.2` and `v3.0.0-rc.3` `depthai-core` default build was using the system (`brew`) installed `xtensor`.

    - _Note: The `v3.0.0-rc.2` and `v3.0.0-rc.3` `depthai-core` cmake project itself was somehow preferring the system installed `xtensor`. I did-not make any changes to `depthai-core` regarding which `xtensor` to use._

3. **`C++17` vs `C++20` Check.** 

    - My findings at that time were (1) the `depthai-core` cmake project appeared to prefer the system `xtensor` and (2) internet searches for the build errors indicated that `C++20` would support the newer `xtensor v0.27.0` code.
    - Base on these findings, I explicity modified `set(CMAKE_CXX_STANDARD 20)` in the `depthai-core` project. This single change resolved all the `xtensor v0.27.0` related build errors.