luxonis/depthai-core #1360: [BUG] Significant drift when using VIO

viggy96, 2025-06-14:
When using Basalt VIO, the pose estimate drifts significantly even when the device is still. 

How can I improve the behavior of VIO? 

I need accurate VIO for my application. 

I have seen this behavior on both the OAK-D Pro W, and the OAK-D Lite (w/IMU).

moratom, 2025-06-15:
Hi @viggy96, thanks for testing it out!

We've merged a critical change to develop after the `rc.2` went out, so first thing to do would be to test on `develop`.

I assume you were using `rc.2`?

viggy96, 2025-06-15:
Correct, I'm using rc.2, how would I go about testing develop?

moratom, 2025-06-15:
Are you using Python or C++?

moratom, 2025-06-15:
For Python you can do `git checkout develop && git pull` on this repository and then do `python3 examples/python/install_requirements.py` which will install the correct version for you.

Or you can just run the command below for the current develop:
```
python3 -m pip install -U --prefer-binary --user --extra-index-url https://artifacts.luxonis.com/artifactory/luxonis-python-snapshot-local depthai==3.0.0-rc.2.dev0+598c1139f42e3d78de4d12f8bf8d832f7a2c8105
```

> I'm using Python. What critical change are you referring to, that you suspect will result in an improvement?
We unified coordinate systems for the IMUs and missed a change that needed to be done in BasaltVIO node.
With it in place, the drfit should go away.

Out of curiousity, what do you plan to use the VIO for? We'd appreciate any furhter feedback.

viggy96, 2025-06-15:
I plan on using it for FRC, FIRST Robotics Competition, which is a competition of high school robotics teams run by FIRST, a worldwide STEM non-profit, whose aim is to get young people excited about STEM fields, and give them real life skills to be better people. 

In matches, it is often useful to know where the robot is on the field, such that the robot can automate certain actions when in a certain area, or to drive autonomously. 

It would further help if AprilTags could be integrated, as anchor points, so that when the device sees AprilTags, it will correct its position. Preferably this AprilTag feature would automatically reject tags whose orientation appears to be incorrect, based on the camera's own orientation relative to the gravity direction, and the known orientation of the tag relative to gravity.

viggy96, 2025-06-15:
Hmm, actually I do still see significant drift, upon further testing.

viggy96, 2025-06-15:
Yeah Basalt still drifts significantly

viggy96, 2025-06-16:
Possibly, though as mentioned earlier, it loses tracking a lot, with any sudden movement. So it's not really suitable for our use case.

viggy96, 2025-06-23:
For reference, here is the exact code I ran to produce this log. You may uncomment the print line to see the pose data, and its drift. Also, the above log was produced with the device sitting motionless, face up on a table.

```
#!/usr/bin/env python

import signal
import time
import depthai as dai

from wpimath.geometry import Pose3d, Translation3d, Rotation3d, Quaternion

# Create pipeline
with dai.Pipeline() as p:
    fps = 60
    width = 640
    height = 400
    # Define sources and outputs
    left = p.create(dai.node.Camera).build(dai.CameraBoardSocket.CAM_B, sensorFps=fps)
    right = p.create(dai.node.Camera).build(dai.CameraBoardSocket.CAM_C, sensorFps=fps)
    imu = p.create(dai.node.IMU)
    odom = p.create(dai.node.BasaltVIO)

    imu.enableIMUSensor([dai.IMUSensor.ACCELEROMETER_RAW, dai.IMUSensor.GYROSCOPE_RAW], 200)
    imu.setBatchReportThreshold(1)
    imu.setMaxBatchReports(10)

    # Linking
    left.requestOutput((width, height)).link(odom.left)
    right.requestOutput((width, height)).link(odom.right)
    imu.out.link(odom.imu)
    transform_queue = odom.transform.createOutputQueue()

    p.start()
    while p.isRunning():
        transform_message = transform_queue.get()
        temp_point = transform_message.getTranslation()
        temp_quaternion = transform_message.getQuaternion()

        pose = Pose3d(
            Translation3d(temp_point.x, temp_point.y, temp_point.z),
            Rotation3d(Quaternion(temp_quaternion.qw, temp_quaternion.qx, temp_quaternion.qy, temp_quaternion.qz))
        )

        #print(str(pose))
        time.sleep(0.01)
```

moratom, 2025-06-30:
Hi @viggy96 - at this time we're providing the SLAM examples as a starting point, but can't spend the resources for full support yet.

The good news is that the majority of the implementation is completly open source, so if you want you're free to poke around and try to improve the accuracy of the examples.
The relevant implementation is here https://github.com/luxonis/depthai-core/blob/main/src/rtabmap/RTABMapSLAM.cpp

viggy96, 2025-06-30:
I would appreciate any pointers as to what part of the Basalt algorithm is going wrong. I believe you guys were able to replicate the drift issue?

Serafadam, 2025-07-01:
Hi, thanks for the additional logs, as we mentioned this issue is not deterministic so it might be hard to debug correctly, we will be looking into it more in depth, what could be also useful is if you could share some additional information, for example:
- What hardware are you running on
- What system
- Is this happening both on C++ and Python versions
- Could you provide some recordings so that we can see if the cause might be related to environment, for example if there are large shifts in exposure or some other external influences

viggy96, 2025-07-01:
@Serafadam 

I am currently using a OAK-D Pro W.

As far as my system specs, it is a Ryzen 9 5950X, 64GB RAM, 7900XTX GPU, running Manjaro kernel 6.15.

So far I've only tested Python, and I see the issue there, I can try to test on C++.

I will provide recordings of the passthrough as soon as I can.

viggy96, 2025-07-01:
We also see similar behavior on an OAK-D Lite.

viggy96, 2025-07-01:
@Serafadam 
Here is a video from the left camera:

https://github.com/user-attachments/assets/c70538e7-2f3c-4a3a-b1cd-3e2bc90f5de0

viggy96, 2025-07-03:
As a small tangent it is not clear in the documentation how to build a Python wheel with Basalt included. I was able to build DepthAI v3 as a shared library for development purposes, and use it in Python, but whenever I built a wheel, the BasaltVIO node was missing.

En1gma02, 2025-07-26:
> As a small tangent it is not clear in the documentation how to build a Python wheel with Basalt included. I was able to build DepthAI v3 as a shared library for development purposes, and use it in Python, but whenever I built a wheel, the BasaltVIO node was missing.

Facing the same issue, BasaltVIO node is missing, we are trying to perform VIO on Oak-D-Lite camera, can you please suggest how did you built it to execute Basalt in your PurpleRanger project? Thanks!
    odom = p.create(dai.node.BasaltVIO)
                    ^^^^^^^^^^^^^^^^^^
AttributeError: module 'depthai.node' has no attribute 'BasaltVIO'

viggy96, 2025-07-27:
Check out this issue: 

https://github.com/luxonis/depthai-core/issues/1383

PR #1376 (2025-07-14): Use correct extrisics matrix for Basalt
## Purpose
Fixes #1360.

## Specification
None / not applicable

## Dependencies & Potential Impact
None / not applicable

## Deployment Plan
None / not applicable

## Testing & Validation
Tested on an OAK-D Pro W. Device was motionless on a desk, and the resulting pose output was stable for a long time.

## Notes
It appears that the `calib.T_i_c` needs a camera to IMU extrinsics matrix, however an IMU to camera was provided previously.