luxonis/depthai-core #1385: [BUG] [Regression] Severe performance degradation in Basalt VIO

viggy96, 2025-07-15:
**Check if issue already exists**

- Google it (e.g. *error xy* github luxonis depthai)
- Check [troubleshooting](https://docs.luxonis.com/en/latest/pages/troubleshooting/) in documentation.

**Describe the bug**
Basalt VIO RTab SLAM performance has degraded severely, producing a pose estimate once every 4 seconds or so, as opposed to many times per second (probably around 60Hz or so) before.

Reproducible using example code.

viggy96, 2025-07-16:
This issue was not visible on my fork.

moratom, 2025-07-16:
Hi @viggy96 - this is highly likely because we also started to build the dependencies in Debug mode by default.

Try building in Release mode and your issue will likely go away - you can do so like so:
```
-DCMAKE_BUILD_TYPE=Release
```

viggy96, 2025-07-16:
Awesome, looks like that was it, thanks @moratom.

closed 2025-07-16 by viggy96