luxonis/depthai-core #1407: [Feature-Request] Add `getTransformation()` to `Tracklets` for consistent coordinate system handling

richard-xx, 2025-08-01:
### Start with the `why`:
We often transform frames (rotate/crop/resize) before ObjectTracker, but Tracklets has no way to retrieve the same matrix that `ImgFrame`/`ImgDetections` expose via `getTransformation()`. This forces manual re-mapping between coordinate systems, which is error-prone.

### Move to the `what`:
Add `getTransformation()` to `Tracklets`, returning the same matrix. If the input message already contains a matrix, the tracker should pass it through unchanged; otherwise it returns identity.

### Move to the `how`:
If you already have an idea of how this new feature could be implemented, describe it here.

moratom, 2025-08-01:
Hey @richard-xx - that makes sense, we'll add it for the next release, thanks for the feature request!

moratom, 2025-08-05:
@richard-xx we've added this to develop (thanks @asahtik !).

closed 2025-08-05 by moratom