luxonis/depthai-core #1437: [Feature-Request] Add the possibility to configure poxy/certificates

LaurentLouf, 2025-09-02:
### Corporate users are usually behing proxies

I think it is self-explanatory : some users of the Luxonis hardware have computer that are behind corporates proxies, so running the last installation step `python .\examples\python\install_requirements.py` fails miserably in that case, since it tries to access the address directly, bypassing the proxy, which doesn't work

### Proxy/certificate cannot be provided

There may be other instances, but at least the most frustrating is at installation. As the saying goes, you only have one shot at making a good impression, so debugging something as simple as downloading a few files is frustrating. 

Getting back to install_requirements.py, some arguments are available : 
```python
convert_default = "empty"
parser = argparse.ArgumentParser()
parser.add_argument('-sdai', "--skip_depthai", action="store_true", help="Skip installation of depthai library.")
parser.add_argument('-dr', "--dry_run", action="store_true", help="Print commands without executing.")
parser.add_argument("--convert", nargs="?", default=convert_default, help="Convert the NN blobs using BlobConverter. Can be used as --convert 2021.4 to convert using OpenVINO 2021.4 or just --convert to use latest OpenVINO release")
parser.add_argument('-irr', "--install_rerun", action="store_true", help="Install rerun library.")
parser.add_argument('-io3dcpu', "--install_open3d_cpu", action="store_true", help="Install open3d with CPU support.")
```

Though none make it possible to configure proxy/CA. Of course, one can go the docs for _requests_, once you have figured that downloads are performed through it, and set a couple of environment variables, but that isn't very straightforward

### Offer the possibility to add proxy/CA

My quick and dirty fix was to hardcode the value for proxy and CA directly into the _ThreadSessionFactory_ : 
```python
# There is no evidence that the requests.Session class is thread-safe,
# so for safety, we use one Session per thread. This class ensures that
# each thread gets its own Session.
class ThreadSessionFactory:
    def __init__(self, exit_stack):
        self._lock = threading.Lock()
        self._thread_local = threading.local()
        self._exit_stack = exit_stack

    def __call__(self):
        try:
            session = self._thread_local.session
        except AttributeError:
            with self._lock: # ExitStack might not be thread-safe either
                session = self._exit_stack.enter_context(requests.Session())
            session.proxies.update({'http': 'http://127.0.0.1:9000', 'https': 'http://127.0.0.1:9000'})
            session.verify = "C:\\Users\\LL767856\\Downloads\\luxonis-com-chain.pem"
            self._thread_local.session = session
        return session
```

That isn't very clean, but at least if someone gets there with the same problem, it's a fix that works.

A cleaner approach would be to add this to the list of arguments in the _examples/python/downloader/downloader.py_ file, as _install_requirements.py_ calls it directly (thus, those arguments also need to be added to _install_requirements.py_)

```python
downloader_cmd = [sys.executable, f"{examples_dir}/downloader/downloader.py", "--all", "--cache_dir", f"{examples_dir}/downloader/", "--num_attempts", "5", "-o", f"{examples_dir}/models"]
```
could become 
```python
downloader_cmd = [sys.executable, f"{examples_dir}/downloader/downloader.py", "--all", "--cache_dir", f"{examples_dir}/downloader/", "--num_attempts", "5", "-o", f"{examples_dir}/models", "--proxy", args.proxy, "--ca", args.ca]
```

LaurentLouf, 2025-09-02:
So now I'm trying to get a model running, which of course needs to download the model weights from the Internet. But funny thing is, now we're down to the C++ level. 

For those stumbling into the same problem, this is where I'm at right now : 

- src/modelzoo/Zoo.cpp is where it happens
- from there, you can derive the following url to download a model https://easyml.cloud.luxonis.com/models/api/v1/models/download?slug=yolov6-nano&platform=RVC2
- and now I'm trying to find where I should put the downloaded file, I have `static fs::path modelZooDefaultCachePath = ".depthai_cached_models";  // hidden cache folder` and `static fs::path modelZooDefaultModelsPath = "depthai_models";         // folder` but that doesn't tell me relative to where

moratom, 2025-09-03:
Hey @LaurentLouf - thanks for the suggestions, we aren't yet actively supporting the "main" flow to work without general internet access.

Internet access in runtime if ofcourse not the only way to go.
You can download the model manually from the Zoo and then use the `setNNArchive` API or pass the downloaded NNArchive to the `.build` functions of neural network nodes.

You can create the NNArchive itself like so:
```
archive = dai.NNArchive(archivePath)
```

and then use the `NeuralNetwork.build` overloads that support it as an argument.

LaurentLouf, 2025-09-04:
@moratom well I have general Internet access, it's just behind a proxy and the different websites you're using are not blacklisted. 

It's of course your choice to implement it or not, I just want to emphasize how much of a bad user experience it is to have to spend a few hours getting into your code just to run examples. I'm used to have to configure git and pip to use my proxy, but digging through code just to configure a proxy is a lot more annoying